siemens/meta-iot2000 #142: is chmod avaible in the windows powershell with docker?

IoTcreator, 2019-11-12:
Hello there!

Is `chmod` avaible in the windows powershell with docker? 
If i wanted to build kas-docker
( doing this obviously before: `git clone https://github.com/siemens/meta-iot2000` 
and 
`wget https://raw.githubusercontent.com/siemens/kas/master/kas-docker`)
`chmod a+x kas-docker`
it shows me the error that `chmod` isnt found in the windows powershell with docker running. 
Hoping for help. 
Best regards!

jan-kiszka, 2019-11-12:
Never tried on windows, but you will need a POSIX compatible shell for the kas-docker script plus some additional tools.

Alternative: Try to invoke the kas docker container interactively and run inside the container.

    docker run -it kasproject/kas

And then follow the instructions for the container-free build. Note that the build result, the image, will have to be copied out afterwards in order to flash it to the SD card under Windows.

IoTcreator, 2019-11-12:
got it! Worked fine! 
Thank you very much for the quick solution!

closed 2019-11-12 by IoTcreator