siemens/meta-iot2000 #146: usb-ser driver ch341 outdated

df5zx, 2020-01-08:
On testing a program with a serial device, it was running fine with PL210x- and ftdi-USB-serial adapters plugged into the IOT, it failed with wrong received data on two CH34x-based adapters. Practically the same program under WINDOWS was working OK with all four adapters.

Searching for the cause I found that in older LINUX distros a broken CH340 driver was included. Could you please update the ch341.c with ch34x.c, from here: [ juliagoda /
CH341SER](https://github.com/juliagoda/CH341SER) .

Regards,
Jürgen

jan-kiszka, 2020-01-09:
That link points to a downstream fork. Does drivers/usb/serial/ch341.c from current mainline work? Then you can compile a custom image with either a recent kernel (there is linux-mainline-latest.bb which may serve as a baseline) or enhance 4.4-cip on top of this layer with patches from `git log --oneline  v4.4..v5.5-rc5 drivers/usb/serial/ch341.c`
```
35714565089e USB: serial: ch341: reimplement line-speed handling
e33eab9ded32 USB: serial: ch341: fix type promotion bug in ch341_control_in()
627cfa89b1fe USB: serial: fix module-license macros
6ca98bc2843e USB: serial: Remove redundant license text
5fd54ace4721 USB: add SPDX identifiers to all remaining files in drivers/usb/
7c61b0d5e82b USB: serial: ch341: change initial line-control settings
448b6dc5a964 USB: serial: ch341: rename LCR variable in set_termios
e8024460354c USB: serial: ch341: rename modem-status register
beea33d4f94b USB: serial: ch341: rename shadow modem-control register
91e0efcd4c23 USB: serial: ch341: clean up control debug messages
a0467a967f34 USB: serial: ch341: fix modem-status handling
2d5a9c72d0c4 USB: serial: ch341: fix control-message error handling
55fa15b5987d USB: serial: ch341: fix baud rate and line-control handling
3cca8624b662 USB: serial: ch341: fix line settings after reset-resume
ce5e29282811 USB: serial: ch341: fix resume after reset
f2950b78547f USB: serial: ch341: fix open error handling
030ee7ae52a4 USB: serial: ch341: fix modem-control and B0 handling
a20047f36e2f USB: serial: ch341: fix open and resume after B0
4e2da44691cf USB: serial: ch341: fix initial modem-control state
a98b69002a16 USB: serial: ch341: add debug output for chip version
ba781bdf8662 USB: serial: ch341: add support for parity, frame length, stop bits
4e46c410e050 USB: serial: ch341: reinitialize chip on reconfiguration
6fde8d29b042 USB: serial: ch341: add register and USB request definitions
fb571101af63 USB: serial: fix compare_const_fl.cocci warnings
```

jan-kiszka, 2021-03-02:
Resolved?