siemens/meta-iot2000 #147: Building failed due to node red recipe's functions

veronicapr, 2020-01-10:
I have followed the instructions in the README.md file, however, when I run the "kas build" command, I get errors on some of five of the node red related tasks.

The errors occur in the following tasks:
- /meta-iot2000/meta-iot2000/meta-iot2000-example/recipes-node/mindconnect-node-red-contrib-mindconnect/mindconnect-node-red-contrib-mindconnect_3.5.0.bb:do_package
-   /meta-iot2000/meta-iot2000/meta-iot2000-example/recipes-node/node-red-contrib-modbus/node-red-contrib-modbus_4.1.3.bb:do_package
-   /meta-iot2000/meta-iot2000/meta-iot2000-example/recipes-node/node-red-contrib-iiot-opcua/node-red-contrib-iiot-opcua_3.10.2.bb:do_package
-   /meta-iot2000/meta-iot2000/meta-iot2000-example/recipes-node/node-red/node-red_0.20.7.bb:do_package
-   /meta-iot2000/meta-iot2000/meta-iot2000-example/recipes-node/node-red/node-red_0.20.7.bb:do_populate_sysroot

I attached the error logs of the 5 failed tasks.

[image]
[image]
[image]
[image]
[image]

How can I solve this?

Thanks in advance.

Best Regards,
Veronica

jan-kiszka, 2020-01-11:
Hmm, there must be something in the host system that triggers this. If I get this correctly, you are building natively on a Linux system (not inside a docker container), right? What is the host system, which Linux distro? Did you already try the kas-docker path as an alternative?

veronicapr, 2020-01-13:
Yes, I am building on  Ubuntu Xenial 16.04.6 LTS. I'm not sure what could be triggering this. I also tried to use de kas-docker path but I get the following error:
`standard_init_linux.go:211: exec user process caused "exec format error"`

jan-kiszka, 2020-01-13:
The kas-docker issue might be similar to https://stackoverflow.com/questions/57366007/standard-init-linux-go211-exec-user-process-caused-no-such-file-or-directory - was Windows involved at some stage of the checkout?

veronicapr, 2020-01-13:
It is the same error, but not the same issue because I did the git checkout on ubuntu. I've checked and I have no "\r\n" in my .sh and .bash files.

jan-kiszka, 2020-01-13:
OK, so already starting docker fails, or that error later in the build? Does `docker run -it --rm kasproject/kas:2.0` work?

veronicapr, 2020-01-13:
No, I get the same error. `standard_init_linux.go:211: exec user process caused "exec format error"`

Could it be because I'm using a 32 bit ubuntu?

jan-kiszka, 2020-01-13:
Uuh, yes. 32-bit is dead, and our build environment is 64-bit as well. In theory, a native build (outside docker) on a 32-bit system should work, but it is surely untested.

veronicapr, 2020-01-17:
I tried in a 64 bit system and it worked. Thank you so much!

closed 2020-01-17 by jan-kiszka