siemens/meta-iot2000 #150: Nodejs Error

DannyMillw, 2020-02-23:
Could anyone advise on this build error please?

ERROR: nodejs-8.16.0-r0 do_compile: oe_runmake failed
ERROR: nodejs-8.16.0-r0 do_compile: Execution of '/home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/temp/run.do_compile.42346' failed with exit code 1:
make -C out BUILDTYPE=Release V=1
  touch d725b580133475b12ffcf5a6710e438fee2c6d67.intermediate
  LD_LIBRARY_PATH=/home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/lib.host:/home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/lib.target:$LD_LIBRARY_PATH; export LD_LIBRARY_PATH; cd ../deps/v8/src/inspector; mkdir -p /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj/gen/src/inspector/protocol /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj/gen/include/inspector; python ../../third_party/inspector_protocol/CodeGenerator.py --jinja_dir ../../third_party --output_base "/home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj/gen/src/inspector" --config inspector_protocol_config.json
  i586-poky-linux-g++  -m32 -march=i586 -Wa,-momit-lock-prefix=yes -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/recipe-sysroot -pthread -rdynamic -m32 -Wl,--whole-archive,/home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj.target/libnode.a -Wl,--no-whole-archive -Wl,--whole-archive,/home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj.target/deps/uv/libuv.a -Wl,--no-whole-archive -Wl,-z,noexecstack -Wl,--whole-archive /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj.target/deps/v8/src/libv8_base.a -Wl,--no-whole-archive -pthread -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -o /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/node -Wl,--start-group /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj.target/node/src/node_main.o /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj.target/libnode.a /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj.target/deps/v8/src/libv8_libplatform.a /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj.target/deps/http_parser/libhttp_parser.a /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj.target/deps/cares/libcares.a /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj.target/deps/uv/libuv.a /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj.target/deps/nghttp2/libnghttp2.a /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj.target/deps/v8/src/libv8_base.a /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj.target/deps/v8/src/libv8_libbase.a /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj.target/deps/v8/src/libv8_libsampler.a /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj.target/deps/v8/src/libv8_builtins_setup.a /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj.target/deps/v8/src/libv8_builtins_generators.a /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/obj.target/deps/v8/src/libv8_nosnapshot.a -lz -lcrypto -lssl -licui18n -licuuc -licudata -ldl -lrt -lm -Wl,--end-group
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
node.target.mk:158: recipe for target '/home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/node' failed
make[1]: *** [/home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/node-v8.16.0/out/Release/node] Error 1
rm d725b580133475b12ffcf5a6710e438fee2c6d67.intermediate
Makefile:88: recipe for target 'node' failed
make: *** [node] Error 2
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/danny/build/tmp/work/i586-nlp-32-poky-linux/nodejs/8.16.0-r0/temp/log.do_compile.42346

jan-kiszka, 2020-02-24:
The linker was killed by someone else - maybe you ran out of memory?

What is your build environment? What build method are you using (native or docker)?

DannyMillw, 2020-02-24:
My build method is a Ubuntu 18.04 LTS virtual machine. When I first ran the build I ran out of disk space on the virtual machine, however I have now expanded the disk size. Could this be a issue? should I start a fresh build?

jan-kiszka, 2020-02-24:
Yes, disk size can also be an issue, but this should be reported clearly. Make sure the VM has as much RAM as possible on your host.

DannyMillw, 2020-03-02:
Thanks for your help Jan, increasing the memory in the VM seems to have solved the issue. Thank you

closed 2020-03-05 by DannyMillw