siemens/meta-iot2000 #151: Question: Communication between Linux and Arduino

brammahh, 2020-03-08:
This might not be the right place to ask, but this repo seems to be the most active community around the IOT2000.

The IOT2000 features an onboard Arduino I/O and actually works with Arduino IDE, which is great. What I can't seem to figure out, however, is how to communicate between the Arduino and the Linux host system.

 I've managed for example to hook up an LCD to the Arduino using i2c and display text on there. I haven't found a way to display text from the Linux component, be it NodeRED, a native app or from a shell. My question basically boils down to two things:

- Is there a way for the Arduino code to communicate with the Linux host system? For example, send a string to a Unix socket in Linux, to which the Arduino program listens and then sends onto the i2c bus?

- Is there a way for the Linux host system to directly access the Arduino I/O? Would it be possible to write a program running in Yocto which accesses the i2c bus, the digital and analog pins or the serial bus on the Arduino?

My end goal is to communicate with an external transceiver over a serial connection. This could work by Linux directly accessing the Arduino I/O, or by uploading an Arduino sketch which communicates the data to the host operating system. Any thoughts?

Thank you for actively maintaining the IOT2000 software!

jan-kiszka, 2020-03-08:
I would not recommend building a more complex application on top of the Arduino interface. It's rather fragile and hard to maintain, not to speak of many of the libraries that come from the Arduino world. I'm also not aware that Intel (from where we inherited that Arduino emulation) has foreseen a proper interface to the Linux host.

Have a look at the mraa library and its API. It provides native access to the I/Os and buses in C++, Python or NodeJS. It's also the more sustainable approach because it will survive the cleanup done with the next device generation.

bramroelandts, 2020-03-08:
Thanks a lot for your reply. It’s good to know that the Arduino-route is not the way to go. I have used `mraa`abefore but hadn’t realized it goes way beyond just controlling the user LED and button.

I’m excited for the next generation of the IOT2000. Keep up the good work!

closed 2020-03-08 by bramroelandts