siemens/meta-iot2000 #152: Enabling CONFIG_TUN

DannyMillw, 2020-03-12:
Good evening,
Could anyone assist me with a build error when trying to build with the CONFIG_TUN=y. I am modifying "meta-iot2000-bsp/recipes-kernel/Linux/configs/deconfig to achieve this.

This is the error message I receive when trying to build the image - 
ERROR: iot2000-example-image-1.0-r0 do_rootfs: The postinstall intercept hook 'update_gio_module_cache' failed, details in /home/danny/build/tmp/work/iot2000-poky-linux/iot2000-example-image/1.0-r0/temp/log.do_rootfs

If you would like me to send or post the log file please contact me.

Thank you

jan-kiszka, 2020-03-13:
That log.do_rootfs may tell us more.

FWIW, I've queued CONFIG_TUN=m in https://github.com/siemens/meta-iot2000/commit/7ecf4f3a08cf23dc1ba32c17094910ec855a3ad9.

jan-kiszka, 2021-03-02:
Merged into master.

closed 2021-03-02 by jan-kiszka