siemens/meta-iot2000 #153: Installing NodeJS 10 or above

olivierpichon, 2020-04-15:
Hello there!

We tried to compile Node 10 from source and it failed after building Torque (we got `Illegal instruction`).

We noticed this past commit on this repo:

https://github.com/siemens/meta-iot2000/commit/eec8f87114d371819b15cde921b9ff8c0ec75052

Is any higher version than Node 8 still not supported?

If not, we can't seem to find a package of Node 8 already built (.ipk) file? There is no other way than compiling it from source? The last ipk file we were able to find is http://iotdk.intel.com/repos/iot-cloud-dev/iotdk/3.5/i586/nodejs-dev_v4.4.3-r1.0_i586.ipk

Thanks a lot for your help!

jan-kiszka, 2020-04-15:
NodeJS requires patches in order to enable support for the Quark processor (i.e. an SSE-free 586-class CPU). We have those for the 8 series as encoded in this layer, but work for 10 only started. It's not trivial - again - because a lot changed in the V8 engine between the versions used in NodeJS 8 and 10, and those changes unfortunately affect the processor support as well. I cannot provide an estimate yet when this can be finished.

Note that the Intel repo is dead for a long time now. It only provided usable packages for the very first IOT2000 image version and was then already abandoned by Intel.

olivierpichon, 2020-04-15:
Thanks @jan-kiszka !

So if this Intel repo is dead, which repo should we use to get Node8? Is there any repo we could use to pull it from? Alternatively, is there a guide we could use to install Node8 from source? On the poky-iot2000 v2.6.0 image example we installed, it does not seem to come with it and as stated before, compiling it (v8.16.2) fails.

Thanks again!

jan-kiszka, 2020-04-15:
Node JS 8.16.0 comes with the example image 2.6.0, see https://github.com/siemens/meta-iot2000/tree/master/meta-iot2000-example/recipes-devtools/nodejs for the related recipe.

olivierpichon, 2020-04-15:
Thanks @jan-kiszka

closed 2020-04-15 by olivierpichon