siemens/meta-iot2000 #155: No .wic file generated

Valle342, 2020-07-13:
Hi, although I run the `./kas-docker build meta-iot2000/kas-example.yml` successfully, I don't get the wic-file you talk about in the readme. What am I missing?

deploy-folder looks like this:

deploy/iot2000
`├── bzImage -> bzImage--4.4.208-cip41-r0-iot2000-20200713073846.bin
├── bzImage--4.4.208-cip41-r0-iot2000-20200713073846.bin
├── bzImage-iot2000.bin -> bzImage--4.4.208-cip41-r0-iot2000-20200713073846.bin
├── efibootguardia32.efi
├── modules--4.4.208-cip41-r0-iot2000-20200713073846.tgz
├── modules-iot2000.tgz -> modules--4.4.208-cip41-r0-iot2000-20200713073846.tgz
├── swupdate-image-iot2000-20200713073846.rootfs.ext4.gz.u-boot
├── swupdate-image-iot2000-20200713073846.rootfs.manifest
├── swupdate-image-iot2000-20200713073846.testdata.json
├── swupdate-image-iot2000.ext4.gz.u-boot -> swupdate-image-iot2000-20200713073846.rootfs.ext4.gz.u-boot
├── swupdate-image-iot2000.manifest -> swupdate-image-iot2000-20200713073846.rootfs.manifest
└── swupdate-image-iot2000.testdata.json -> swupdate-image-iot2000-20200713073846.testdata.json
`

BR

jan-kiszka, 2020-07-13:
Are you sure you didn't accidentally built `kas-example-swupdate.yml`? That config would generate all those swupdate-image artifacts, but not the default `kas-example.yml`.

Valle342, 2020-07-13:
wups, sorry. I accidently copied the wrong folder. 
This is the one for kas-example:

`acpi-upgrades-iot2000.cpio
bzImage
bzImage--4.4.208-cip41-r0-iot2000-20200710122501.bin
bzImage-iot2000.bin
core-image-minimal-initramfs-iot2000-20200710122501.rootfs.cpio.gz
core-image-minimal-initramfs-iot2000-20200710122501.rootfs.manifest
core-image-minimal-initramfs-iot2000-20200710122501.testdata.json
core-image-minimal-initramfs-iot2000.cpio.gz
core-image-minimal-initramfs-iot2000.manifest
core-image-minimal-initramfs-iot2000.testdata.json
efibootguardia32.efi
grub-efi-bootia32.efi
modules--4.4.208-cip41-r0-iot2000-20200710122501.tgz
modules-iot2000.tgz
systemd-bootia32.efi
`

Still no .wic-file

jan-kiszka, 2020-07-13:
Did the build succeed? Or was bitbake reporting any error, specifically regarding the wic target? Maybe you ran out of disk space in the build environment, and therefore the image was not fully created.

Valle342, 2020-07-13:
I had enough disk space. I found these error message in the logs:

`ERROR: upm-2.0.0+gitAUTOINC+5cf20df96c-r0 do_compile: Execution of '/work/build/tmp/work/i586-nlp-32-poky-linux/upm/2.0.0+gitAUTOINC+5cf20df96c-r0/temp/run.do_compile.2916930' failed with exit code 1:`

`{standard input}: Assembler messages:
{standard input}:638862: Warning: end of file not at end of a line; newline inserted
i586-poky-linux-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.`

`ERROR: Task (/work/meta-openembedded/meta-oe/recipes-extended/upm/upm_git.bb:do_compile) failed with exit code '1'`

Valle342, 2020-07-27:
The issue seems to simply has been using an unsupported os, namely centOS8. Now on centOS7 things work fine.

closed 2020-07-27 by Valle342

jan-kiszka, 2020-07-27:
Well, one would expect newer versions to work better, and the kas-docker container is supposed to even abstract away to host OS. Maybe you ran into more strict resource limitations with CentOS8?

The more cores you have, the more memory the build consumes and, thus, has to be accessible inside the docker container. That's at least one issue we've seen on memory-limiting CI setups.