siemens/meta-iot2000 #156: swupdate remote

Valle342, 2020-07-27:
I want to use swupdate remotely on my created system with kas-example-swupdate.yml. 
You state "NOTE: The setup can be adapted to work with backends like, e.g., hawkBit.". How can I change the configuration of swupdate properly?

gylstorffq, 2020-07-27:
An example of using swupdate with hawkbit would be the raspberry example of https://github.com/sbabic/meta-swupdate-boards.
The necessary steps are:
- create a swupdate_%.bbappend
- add a new  defconfig for swupdate, for example, https://github.com/sbabic/meta-swupdate-boards/blob/master/recipes-support/swupdate/swupdate/raspberrypi3/defconfig. The defconfig should contain:

```
...
CONFIG_SURICATTA=y
CONFIG_SURICATTA_HAWKBIT=y
...
```

These config items need additional dependencies.
 
- Adapt the command line/swupate.cfg to start swupdate and connect it to the hawkbit server. 

```
swupdate -u '-t <tenant> -u <url to hawkbit server> -i <device id>'
```

Another command line example to start with surricatta is at https://sbabic.github.io/swupdate/suricatta.html#running-suricatta. For the configuration you can look at https://github.com/sbabic/meta-swupdate-boards/blob/master/recipes-support/swupdate/swupdate/raspberrypi3/swupdate.cfg.

Valle342, 2020-07-29:
Hey gylstorffq,

thank you very much for your answer. If I do this I get following error:

`When reparsing /work/meta-swupdate/recipes-support/swupdate/swupdate_2019.11.bb:do_configure, the basehash value changed from c3fa822557d2f3f408ad85ac869147d659ebcb4db307cf552748653d77386a22 to a97c773a1ccd15c3ebb20d21b44468064e9a49e643a65b5f1c2d34c011bf0eb4. The metadata is not deterministic and this needs to be fixed.`

This error I do get also for `do_install`, `do_package` in the swupdate_2019.11.bb file.
Do you have an idea how to solve this?

jan-kiszka, 2020-07-29:
Did you modified the recipe? The error definitely comes when modifications happen during a build. It may come when bitbake only thinks that happened. In that case, a "-c clean" on the target recipe followed by a rebuild may help.

Valle342, 2020-08-11:
Hey Jan, 
thanks for the response. This helped.

gylstorffq you mentioned, that I need to install some dependencies. When I try to build the new image with changes HAWKBIT and SURICATTA capabilities I get following error:
"fatal error: json-c/json.h: No such file or directory"
But I did add json-c as a dependency by adding IMAGE_INSTALL_append = " json-c" to /meta-iot2000-example/conf/distro/poky-iot2000.conf.
Any idea how to solve this issue?

gylstorffq, 2020-08-11:
Hey Valle342,

you need to add the [dependency](http://www.embeddedlinux.org.cn/OEManual/recipes_depenencies.html) to the swupdate recipe with:
```
DEPENDS += "json-c"
```
Other dependencies are added in a similar way.

Valle342, 2020-09-08:
Ok, thanks. I added the line to my swupdate_%.bbappend file. Now I get a quite big wall of text printed and two errors, that are not clear to me. This is a little chunk of it, that might have helpful information:

`| ERROR: oe_runmake failed`
`| WARNING: exit code 1 from a shell command.`
`| ERROR: Execution of '/work/build/tmp/work/i586-nlp-32-poky-linux/swupdate/2019.11-r0/temp/run.do_compile.9766' failed with exit code 1:`
`| make -f /work/build/tmp/work/i586-nlp-32-poky-linux/swupdate/2019.11-r0/git/Makefile silentoldconfig`
`| make -f scripts/Makefile.build obj=scripts/basic`
`| mkdir -p include/linux include/config`
`| make -f scripts/Makefile.build obj=scripts/kconfig silentoldconfig`
`| mkdir -p include/config include/generated`
`| scripts/kconfig/conf  --silentoldconfig Kconfig`
`| make -f scripts/Makefile.build obj=core`
`| make -f scripts/Makefile.build obj=handlers`
`| make -f scripts/Makefile.build obj=corelib`
`| make -f scripts/Makefile.build obj=ipc`

jan-kiszka, 2020-09-08:
Please attach the full build log, at least of the affected recipe. You cut right after the interesting error messages.

Valle342, 2020-09-08:
[error_log.txt](https://github.com/siemens/meta-iot2000/files/5187304/error_log.txt)

Ok, sorry. Here the final lines of the log. Thanks a lot :-)

jan-kiszka, 2020-09-08:
@gylstorffq, seen that before?

```
  i586-poky-linux-gcc  -m32 -march=i586 -Wa,-momit-lock-prefix=yes -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/work/build/tmp/work/i586-nlp-32-poky-linux/swupdate/2019.11-r0/recipe-sysroot -Wp,-MD,corelib/.installer.o.d  -Iinclude  -include include/generated/autoconf.h -std=gnu99 -D_GNU_SOURCE -DNDEBUG -D"SWU_VER=KBUILD_STR(2017.11.0)" -DEMBEDDED_LUA_SRC="_binary_swupdate_handlers_lua_start" -Wall -Wshadow -Wwrite-strings -Wundef -Wstrict-prototypes -Wunused -Wunused-parameter -Wunused-function -Wunused-value -Wmissing-prototypes -Wmissing-declarations -Wno-format-security -Wold-style-definition -finline-limit=0 -fno-builtin-strlen -fomit-frame-pointer -ffunction-sections -fdata-sections -fno-guess-branch-probability -funsigned-char -falign-functions=1 -falign-jumps=1 -falign-labels=1 -falign-loops=1 -static-libgcc -fno-unwind-tables -fno-asynchronous-unwind-tables -Os   -D"KBUILD_STR(s)=#s" -c -o corelib/installer.o corelib/installer.c
corelib/channel_curl.c: In function 'channel_map_curl_error':
corelib/channel_curl.c:294:2: error: duplicate case value
  294 |  case CURLE_SSL_CACERT:
      |  ^~~~
corelib/channel_curl.c:293:2: note: previously used here
  293 |  case CURLE_PEER_FAILED_VERIFICATION:
      |  ^~~~
```

Too old swupdate?

gylstorffq, 2020-09-08:
Look like you need a version with commit https://github.com/sbabic/swupdate/commit/58a9997fdac311dda2449228a36edec77f88674a in it. The current commit https://github.com/sbabic/swupdate/commit/012374087c747f05216a7f002e4b8a34bc142c52 in the swupdate file  is too old. 

@Valle342  You can try to set the SRCREV in https://github.com/siemens/meta-iot2000/blob/master/meta-iot2000-example/recipes-support/swupdate/swupdate_%25.bbappend to 58a9997fdac311dda2449228a36edec77f88674a.

Valle342, 2020-09-14:
Guys, thank you. This solved the issue and I could create a .swu-image with hawkbit and suricatta activated. 
But after restarting and changing the partitions, I still get the message: `"swupdate: invalid option -- 'u'"`, when I do `swupdate -f /etc/swupdate/swupdate.cfg -u ''`. I thought, if I activate suricatta in the defconfig, the u-option is available.

gylstorffq, 2020-09-14:
Did you try to use the complete command, e.g. 
```
./swupdate -l 5 -u '-t default -u http://10.0.0.2:8080 -i 25'
```

The u option should be available if surricatta is active.
The following option should be set:
```
...
CONFIG_CURL=y
CONFIG_CHANNEL_CURL=y
CONFIG_SURICATTA=y
CONFIG_SURICATTA_STATE_CHOICE_NONE=y
CONFIG_SURICATTA_HAWKBIT=y
CONFIG_JSON=y
...
```
Can you post your config from swupdate?

Valle342, 2020-09-17:
Hey guys,

ok, now everything is working according to my issue. I missed to activate the CONFIG_SURICATTA_STATE_CHOICE_NONE-config.
Thanks!

Valle342, 2020-09-21:
Hey,

this is probably the wrong place to ask, but I don't know where elsewhere...
If I start the suricatta daemon, I get the error: 
`SSL: certificate subject name '...' does not match target host name '...'`. Now I want to disable this host name verification completely. Do you have an idea how to do this with swupdate?
Thanks in advance!

jan-kiszka, 2020-09-21:
@gylstorffq possibly knows, but you may also ask at swupdate@googlegroups.com.

Valle342, 2020-09-21:
Ok.
Another question related to this topic:
If I restart the device after updating, how can I execute commands after startup (for example start the suricatta daemon and confirm working environment with bg_setenv -c)?

gylstorffq, 2020-09-21:
Does your server provide a http connection or https only? 

The ssl Problem looks like it uses https. You could also add a additional url to your server.

Valle342, 2020-09-22:
The server provides only https and I want to use only https. Why is that relevant?
I am working in a complex VPN-structure, leading to changing IP-Adresses. Additionally DNS is not allowed to use. Therefore I want to disable the whole thing :-/

Ok, how are these services enabled?

gylstorffq, 2020-09-22:
I think for disabling the ssl check you need to ask the swupdate mailing list mailto:swupdate@googlegroups.com.

To activate a service, you would add
ˋˋˋ

SYSTEMD_SERVICE_${PN} = "<Service Name >.service"

ˋˋˋ
To the recipe. For swupdate this should be done already. You need to modify the shell script which is executed by the service. 
Current implementation of the script:
https://github.com/sbabic/meta-swupdate/blob/master/recipes-support/swupdate/swupdate/swupdate.sh

Valle342, 2020-09-29:
Perfect, thanks!

closed 2020-09-29 by Valle342