siemens/meta-iot2000 #157: Error while trying to add node-red-contrib-msg-queue to a custom image

olivierpichon, 2020-07-28:
Hi there,

we followed the process described here to pre-install some node modules to our based image: https://github.node-red-contrib-msg-queuecom/siemens/meta-iot2000/blob/67920485d67829ab02f96297db8e5d8c3b1ebc73/meta-iot2000-example/classes/npm-ng.bbclass#L6

That worked well until now. Trying to do the same to install `node-red-contrib-msg-queue` is failing though with the following error:

```
ERROR: node-red-contrib-msg-queue-1.0.2-r0 do_package_qa: QA Issue: non -staticdev package contains static .a library: node-red-contrib-msg-queue path '/work/i586-nlp-32-poky-linux/node-red-contrib-msg-queue/1.0.2-r0/packages-split/node-red-contrib-msg-queue/home/root/.node-red/node_modules/node-red-contrib-msg-queue/node_modules/sqlite3/build/Release/sqlite3.a'
non -staticdev package contains static .a library: node-red-contrib-msg-queue path '/work/i586-nlp-32-poky-linux/node-red-contrib-msg-queue/1.0.2-r0/packages-split/node-red-contrib-msg-queue/home/root/.node-red/node_modules/node-red-contrib-msg-queue/node_modules/sqlite3/build/Release/obj.target/deps/sqlite3.a' [staticdev]
```

It is worth noting than installing the package directly on the IoT 2040 works. It downloads and compiles a version of sqlite. This process takes more than 45 minutes though and so would be way better for us to pre-build the module in our image. But we have been unsuccessful for now. Any thoughts on how to get that working?

Thanks a lot!

jan-kiszka, 2020-07-28:
Try `INSANE_SKIP_${PN} += "staticdev"`. Probably an acceptable workaround in this scenario.

olivierpichon, 2020-07-28:
Thanks a lot for such a quick feedback @jan-kiszka , we're getting further, now the error is:
```
ERROR: Nothing RPROVIDES 'echo' (but /home/user/yocto/iot2000-enspara/meta-iot2000-enspara/recipes-node/node-red-contrib-msg-queue/node-red-contrib-msg-queue_1.0.2.bb RDEPENDS on or otherwise requires it)
```

I tried to add
`RDEPENDS_${PN} += "echo"`
But it is apparently not a thing 👎

jan-kiszka, 2020-07-28:
Hmm, never had that before. Maybe that thing just want a shell (`RDEPENDS_${PN} += "bash"`)?

olivierpichon, 2020-07-29:
Unfortunately that did not work, I got away with it using:
`INSANE_SKIP_${PN} += "file-rdeps"`
which I bet is pretty unclean...

But anyway you put us on the right track by advising `INSANE_SKIP_${PN} += "staticdev"`
Thanks a lot for your help @jan-kiszka !

closed 2020-07-29 by olivierpichon