siemens/meta-iot2000 #158: building the kas-example image 5 tasks failed, image not created

R0HX, 2020-08-05:
Hello Guys,
 
I was trying to build the kas-example image native via Ubuntu 16.04LTS but in the end 5 tasks failed an I am not able to find the image. I got no idea where these errors come from. I need your help. 

The complete output while building.. 

`jan@jansUbu:~$ kas build meta-iot2000/kas-example.yml 

2020-08-04 13:52:02 - INFO     - kas 2.1.1 started 

2020-08-04 13:52:02 - INFO     - /home/jan/meta-iot2000$ git rev-parse --show-toplevel 

2020-08-04 13:52:02 - INFO     - /home/jan/meta-iot2000$ git rev-parse --show-toplevel 

2020-08-04 13:52:02 - INFO     - Using /home/jan/meta-iot2000 as root for repository meta-iot2000 

2020-08-04 13:52:02 - INFO     - /home/jan$ git clone -q https://git.yoctoproject.org/git/poky /home/jan/poky 

2020-08-04 13:52:02 - INFO     - /home/jan$ git clone -q https://github.com/siemens/meta-efibootguard /home/jan/meta-efibootguard 

2020-08-04 13:52:02 - INFO     - /home/jan$ git clone -q https://github.com/sbabic/meta-swupdate /home/jan/meta-swupdate 

2020-08-04 13:52:02 - INFO     - /home/jan$ git clone -q http://git.openembedded.org/meta-openembedded /home/jan/meta-openembedded 

2020-08-04 13:52:03 - INFO     - Repository meta-efibootguard cloned 

2020-08-04 13:52:04 - INFO     - Repository meta-swupdate cloned 

2020-08-04 13:53:01 - INFO     - Repository meta-openembedded cloned 

2020-08-04 13:53:45 - INFO     - Repository poky cloned 

2020-08-04 13:53:45 - INFO     - /home/jan/meta-iot2000$ git rev-parse --show-toplevel 

2020-08-04 13:53:46 - INFO     - Using /home/jan/meta-iot2000 as root for repository meta-iot2000 

2020-08-04 13:53:46 - INFO     - /home/jan/poky$ git status -s 

2020-08-04 13:53:46 - INFO     - /home/jan/poky$ git rev-parse --verify HEAD 

2020-08-04 13:53:46 - INFO     - 5e1bc3ef5ac83b244102011dc5bfd167a6342514 

2020-08-04 13:53:46 - INFO     - /home/jan/poky$ git checkout -q cf92a2d567260b91a259652bad0ecd790750f710 

2020-08-04 13:53:46 - INFO     - /home/jan/meta-efibootguard$ git status -s 

2020-08-04 13:53:46 - INFO     - /home/jan/meta-efibootguard$ git rev-parse --verify HEAD 

2020-08-04 13:53:46 - INFO     - a802993251b5245c6bb026a3a78464992fa280f6 

2020-08-04 13:53:46 - INFO     - /home/jan/meta-efibootguard$ git checkout -q 3295b62225465b4f792d2c0070420516e2beff45 

2020-08-04 13:53:46 - INFO     - /home/jan/meta-swupdate$ git status -s 

2020-08-04 13:53:46 - INFO     - /home/jan/meta-swupdate$ git rev-parse --verify HEAD 

2020-08-04 13:53:46 - INFO     - e3606b2a58409b8b9f74b076da4263f1b0e520f4 

2020-08-04 13:53:46 - INFO     - /home/jan/meta-swupdate$ git checkout -q 8ad6a75e1778cdda671b3d6aea2a9692613020ec 

2020-08-04 13:53:46 - INFO     - /home/jan/meta-openembedded$ git status -s 

2020-08-04 13:53:46 - INFO     - /home/jan/meta-openembedded$ git rev-parse --verify HEAD 

2020-08-04 13:53:46 - INFO     - 2cf66ec68f32e7fdebaf683ad9b05c399662960f 

2020-08-04 13:53:46 - INFO     - /home/jan/meta-openembedded$ git checkout -q e855ecc6d35677e79780adc57b2552213c995731 

2020-08-04 13:53:46 - INFO     - /home/jan/meta-iot2000$ git rev-parse --show-toplevel 

[... 254 lines cut ...]
ERROR: Task (virtual:native:/home/jan/poky/meta/recipes-devtools/opkg/opkg_0.4.1.bb:do_fetch) failed with exit code '1' 

NOTE: Tasks Summary: Attempted 4610 tasks of which 2 didn't need to be rerun and 5 failed. 

 

Summary: 5 tasks failed: 

  virtual:native:/home/jan/poky/meta/recipes-support/icu/icu_64.2.bb:do_fetch 

  /home/jan/poky/meta/recipes-support/icu/icu_64.2.bb:do_fetch 

  /home/jan/poky/meta/recipes-devtools/opkg/opkg_0.4.1.bb:do_fetch 

  /home/jan/meta-openembedded/meta-oe/recipes-extended/haveged/haveged_1.9.6.bb:do_fetch 

  virtual:native:/home/jan/poky/meta/recipes-devtools/opkg/opkg_0.4.1.bb:do_fetch 

Summary: There were 12 WARNING messages shown. 

Summary: There were 11 ERROR messages shown, returning a non-zero exit code. `

If anything else is needed please let me know! 
Thanks in advance 
BR

jan-kiszka, 2020-08-05:
Use a public (non-corporate, non-filtered) Internet connection.

R0HX, 2020-08-12:
Got it, thank you! But got another Problem after trying to compile the image via bitbake with a bluetooth layer added. 

`ERROR: Error importing OE modules: module 'bb.siggen' has no attribute 'SignatureGeneratorUniHashMixIn'
ERROR: Invalid signature generator 'OEBasicHash', using default 'noop'
Available generators: basic, basichash, noop
ERROR: Invalid signature generator 'OEBasicHash', using default 'noop'
Available generators: basic, basichash, noop
ERROR: Error importing OE modules: module 'bb.siggen' has no attribute 'SignatureGeneratorUniHashMixIn'
ERROR: Invalid signature generator 'OEBasicHash', using default 'noop'
Available generators: basic, basichash, noop
ERROR: Invalid signature generator 'OEBasicHash', using default 'noop'
Available generators: basic, basichash, noop
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    Bitbake version 1.43.2 is required and version 1.36.0 was found
`
Again I have no idea why Bitbake isn't the latest version. I installed just a view days ago the master of poky... 

Thank you in advance 
BR

jan-kiszka, 2020-08-12:
Are you building current master? That is Yocto 3.0-based and comes with bitbake 1.44. Our last release is... older.

R0HX, 2020-08-18:
HI Jan, im so sorry for answering that late all the time. Last time I just messed up something, after rebuilding with master it worked better, but unfortunately there is a new error which gets me again a headache. Again I'm sorry for my late replys. 

> Loading cache: 100% |######################################################| Time: 0:00:01
Loaded 3410 entries from dependency cache.
ERROR: ParseError at /home/ctree/meta-iot2000/meta-iot2000-bsp/recipes-kernel/linux/linux-cip_4.4.bb:12: unparsed line: 'KMACHINE_iot2000 = "intel-quark"s'
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

Thank you in advance 
BR

jan-kiszka, 2020-08-18:
Please describe your build environment in more details (native or kas-docker, Linux distro used, build command issued).

R0HX, 2020-08-19:
I'm using a virtual machine with ubuntu 16.04.6. I was using the guide from the siemens forum to create a image with bluetooth support. 
[include_BT.pdf](https://github.com/siemens/meta-iot2000/files/5094176/include_BT.pdf)
I followed all steps until the error occurred after I tried `bitbake iot2000-example-image`

BR

jan-kiszka, 2020-08-19:
The instructions under "Build the Image with the added components" are outdated by several releases. Please follow the README of this layer, it's in sync with what is required.

R0HX, 2020-08-19:
Sorry for the pathetic question. But which layer do you mean? 

BR

jan-kiszka, 2020-08-19:
meta-iot2000 - actually "repo", containing the two layers (bsp and example). They have external deps which are resolved and configured by kas. Thus call kas to build the example-image.

R0HX, 2020-08-19:
I'm sorry for beeing so totaly out of order right now. But how do I go on? Or is it even a good idea to try to implement bluetooth?

BrotKolben, 2020-08-19:
Hello,
Maybe I can help with this topic: 
@R0HX: first of all, maybe try building the image from scratch using [this guide](https://github.com/siemens/meta-iot2000) as mentioned by jan, without changing any configs. If this works out fine, then we have a good starting point. 
Instead of changing the recipes, maybe its more fail-proof to just add the additional kernel-flags for bluetooth in [this file](https://github.com/siemens/meta-iot2000/blob/master/meta-iot2000-example/recipes-kernel/linux/configs/iot2000-example.cfg) 
```
CONFIG_BT=m
CONFIG_BT_RFCOMM=m
CONFIG_BT_RFCOMM_TTY=y
CONFIG_BT_BNEP=m
CONFIG_BT_BNEP_MC_FILTER=y
CONFIG_BT_BNEP_PROTO_FILTER=y
CONFIG_BT_HIDP=m
CONFIG_BT_HCIBTUSB=m
CONFIG_BT_HCIVHCI=m
```
Then add the respective bluez5 packages in [iot2000-example-image.inc](https://github.com/siemens/meta-iot2000/blob/master/meta-iot2000-example/recipes-core/images/iot2000-example-image.inc) as described, and rebuild the images using the kas script (not bitbake commands like in this old PDF).

Hope this helps. Let us know if everything worked out, then we could include these changes to future images. 

BR
Felipe R.

R0HX, 2020-08-19:
Thank you for your engagment! Executed all steps but unfortunately got the same error again :( 

> ctree@ctree-VirtualBox:~$ kas build meta-iot2000/kas-example.yml
2020-08-19 12:31:23 - INFO     - kas 2.1.1 started
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-iot2000$ git rev-parse --show-toplevel
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-iot2000$ git rev-parse --show-toplevel
2020-08-19 12:31:23 - INFO     - Using /home/ctree/meta-iot2000 as root for repository meta-iot2000
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-openembedded$ git remote set-url origin http://git.openembedded.org/meta-openembedded
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-swupdate$ git remote set-url origin https://github.com/sbabic/meta-swupdate
2020-08-19 12:31:23 - INFO     - /home/ctree/poky$ git remote set-url origin https://git.yoctoproject.org/git/poky
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-efibootguard$ git remote set-url origin https://github.com/siemens/meta-efibootguard
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-openembedded$ git cat-file -t e855ecc6d35677e79780adc57b2552213c995731
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-swupdate$ git cat-file -t 8ad6a75e1778cdda671b3d6aea2a9692613020ec
2020-08-19 12:31:23 - INFO     - /home/ctree/poky$ git cat-file -t cf92a2d567260b91a259652bad0ecd790750f710
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-efibootguard$ git cat-file -t 3295b62225465b4f792d2c0070420516e2beff45
2020-08-19 12:31:23 - INFO     - Repository meta-openembedded already contains e855ecc6d35677e79780adc57b2552213c995731 as commit
2020-08-19 12:31:23 - INFO     - Repository meta-swupdate already contains 8ad6a75e1778cdda671b3d6aea2a9692613020ec as commit
2020-08-19 12:31:23 - INFO     - Repository poky already contains cf92a2d567260b91a259652bad0ecd790750f710 as commit
2020-08-19 12:31:23 - INFO     - Repository meta-efibootguard already contains 3295b62225465b4f792d2c0070420516e2beff45 as commit
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-iot2000$ git rev-parse --show-toplevel
2020-08-19 12:31:23 - INFO     - Using /home/ctree/meta-iot2000 as root for repository meta-iot2000
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-openembedded$ git status -s
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-openembedded$ git rev-parse --verify HEAD
2020-08-19 12:31:23 - INFO     - e855ecc6d35677e79780adc57b2552213c995731
2020-08-19 12:31:23 - INFO     - Repo meta-openembedded has already been checked out with correct refspec. Nothing to do.
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-swupdate$ git status -s
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-swupdate$ git rev-parse --verify HEAD
2020-08-19 12:31:23 - INFO     - 8ad6a75e1778cdda671b3d6aea2a9692613020ec
2020-08-19 12:31:23 - INFO     - Repo meta-swupdate has already been checked out with correct refspec. Nothing to do.
2020-08-19 12:31:23 - INFO     - /home/ctree/poky$ git status -s
2020-08-19 12:31:23 - INFO     - /home/ctree/poky$ git rev-parse --verify HEAD
2020-08-19 12:31:23 - INFO     - cf92a2d567260b91a259652bad0ecd790750f710
2020-08-19 12:31:23 - INFO     - Repo poky has already been checked out with correct refspec. Nothing to do.
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-efibootguard$ git status -s
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-efibootguard$ git rev-parse --verify HEAD
2020-08-19 12:31:23 - INFO     - 3295b62225465b4f792d2c0070420516e2beff45
2020-08-19 12:31:23 - INFO     - Repo meta-efibootguard has already been checked out with correct refspec. Nothing to do.
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-iot2000$ git rev-parse --show-toplevel
2020-08-19 12:31:23 - INFO     - Using /home/ctree/meta-iot2000 as root for repository meta-iot2000
2020-08-19 12:31:23 - INFO     - /home/ctree/poky$ /tmp/tmpjll45yj_ /home/ctree/build
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-iot2000$ git rev-parse --show-toplevel
2020-08-19 12:31:23 - INFO     - Using /home/ctree/meta-iot2000 as root for repository meta-iot2000
2020-08-19 12:31:23 - INFO     - /home/ctree/meta-iot2000$ git rev-parse --show-toplevel
2020-08-19 12:31:23 - INFO     - Using /home/ctree/meta-iot2000 as root for repository meta-iot2000
2020-08-19 12:31:23 - INFO     - /home/ctree/build$ /home/ctree/poky/bitbake/bin/bitbake -k -c build iot2000-example-image
Loading cache: 100% |######################################################| Time: 0:00:01
Loaded 3410 entries from dependency cache.
ERROR: ParseError at /home/ctree/meta-iot2000/meta-iot2000-bsp/recipes-kernel/linux/linux-cip_4.4.bb:12: unparsed line: 'KMACHINE_iot2000 = "intel-quark"s'
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

jan-kiszka, 2020-08-19:
Your meta-iot2000 repository checkout is clean? I can't see a reason for this anymore, and works properly here for current master in a clean checkout.

You may also try the `kas-docker` build variant, to reduce the impact of your rather old host distro on the build.

jan-kiszka, 2021-03-02:
Assuming the issue was resolved as there was no follow-up.

closed 2021-03-02 by jan-kiszka