siemens/meta-iot2000 #166: Problem with Digital input/outputs iot 2040

soportekenai, 2021-03-01:
Hi,
I have an iot2040 with the iot2000 input/output module.
I use the analog current inputs to take sensor readings, this works great.

Also I use digital inputs to read activate buttons and outputs to activate relays, with the followings problems:

We have realized that sometimes after a certain time or when we do some deploy in NodeRed, we begin to have problems with the digital in/outputs. They stop working

It also happens to us with the user led, the red led works but no the green led

Now i am activating the digital outputs by running a python from node-red but it is a problem -> time delay, digital input not work...

I use official SD image Siemens: 
-NodeRed 0.20.7v
-node-red-node-intel-gpio 1.0.0-iot2000

Thanks

jan-kiszka, 2021-03-01:
Maybe the same issue as in https://github.com/siemens/meta-iot2050/issues/57. We fixed it for the 2050, but we did not backport this pattern yet.

Can you describe the error in more details? Does node-red report anything in the log? Does restarting node-red resolve the problem? Or only rebooting the box?

soportekenai, 2021-03-01:
I only use intel gpio module in node red. because node-red-contrib-iot2000 hasn't got modules to use digital outputs and analog inputs.

Node-red log doesn't show no mistake.

The problem is resolved when restarting node-red (sometimes) or rebooting the box

jan-kiszka, 2021-03-01:
I do not mean node-red-contrib-iot2000. I'm always referring to the node-red-nodes-intel-gpio as they are delivered (i.e. ...1.0.0-iot2000) by meta-iot2000 and our example image.

Let me see if I can easily port the fixes we made for meta-iot2050 also to meta-io2000.

soportekenai, 2021-03-01:
Thank you very much, I await your news

jan-kiszka, 2021-03-01:
I've pushed https://github.com/siemens/meta-iot2000/commits/jan/mraa-node-gpio-update, but it is **only build-tested** so far. As patches from the IOT2050 applied rather smoothly, it _should_ work, though.

jan-kiszka, 2021-03-02:
Now also tested, and it appears to work fine.

soportekenai, 2021-03-02:
Perfect, thank you very much. Let's try it and I'll tell you

soportekenai, 2021-03-02:
I need your help because I am not sure how to make the changes on my device

jan-kiszka, 2021-03-02:
You will need to rebuild an image from my branch - or port the commits to your version of meta-iot2000. But there is no way around (re)building an image from sources.

soportekenai, 2021-03-02:
Sorry for my ignorance, but I don't know how to make these changes on my iot2040. We downloaded the OS image but I don't know how to rebuild an image from your branch or port the commits to my version of meta-iot2000. Can I modify the files directly? Thanks

jan-kiszka, 2021-03-02:
You could edit the nodes, but you also need a recompiled libmraa. So, have a look at the README that comes with this repo and try to follow the build instructions, either on a supported Linux distro or inside the docker container.

soportekenai, 2021-03-08:
Hi,

We are trying to follow the steps to build the image but we cannot do it.
 
We need a solution to this problem, 
we need you to explicitly attach the steps to follow or send us the image 
already created. Excuse me for my ignorance

[image]

jan-kiszka, 2021-03-08:
First, install docker or podman on your Linux machine/VM. Make sure you can use it as the logged-in user.

Then:
```
git clone https://github.com/siemens/meta-iot2000
cd meta-iot2000
git checkout jan/mraa-node-gpio-update
wget https://raw.githubusercontent.com/siemens/kas/2.4/kas-container
chmod a+x kas-container
./kas-container build kas-example.yml
```

That said, understanding what went wrong with using kas natively (without a container) would be nice as well. How did you install it? On which distribution/version did you do that?

soportekenai, 2021-03-09:
I have created a Linux 16.04 LTS VM with docker
I have followed your instructions but it gave me error 1
[image]

I have followed the link below to try to solve
https://docs.docker.com/engine/install/linux-postinstall/

But I get the following error 2
[image]

I have followed the link below to try to solve
https://www.digitalocean.com/community/questions/how-to-fix-docker-got-permission-denied-while-trying-to-connect-to-the-docker-daemon-socket

But I get the following error 3
[image]

Wouldn't it be easier for you to generate the image and send me a wetransfer or similar?

jan-kiszka, 2021-03-09:
Please see https://support.industry.siemens.com/tf/ww/en/posts/how-to-build-github-example-image/237116/?page=0&pageSize=10, steps 1-6, for setting up a suitable build VM (not only for the IOT2050).

Regarding a prebuilt image, please contact our support directly.