siemens/meta-iot2000 #167: build error

matejkamarek, 2021-04-29:
Hello,

I'm getting the following compilation error:

`
gcc  -std=gnu99 -DHAVE_CONFIG_H -I. -I../git  -I../git/include -I../git -include config.h -I../git/tools -isystem/data/meta-iot2000/build/tmp/work/x86_64-linux/efibootguard-native/0.7-r0/recipe-sysroot-native/usr/include -Wno-unused-parameter -Wmissing-prototypes -fshort-wchar -DHAVE_ENDIAN_H -D_GNU_SOURCE -D_FILE_OFFSET_BITS=64 -Werror -isystem/data/meta-iot2000/build/tmp/work/x86_64-linux/efibootguard-native/0.7-r0/recipe-sysroot-native/usr/include -O2 -pipe -c -o tools/libebgenv_a-ebgpart.o `test -f 'tools/ebgpart.c' || echo '../git/'`tools/ebgpart.c
In file included from /usr/include/fcntl.h:77:0,
                 from ../git/include/ebgpart.h:46,
                 from ../git/tools/ebgpart.c:19:
/usr/include/bits/stat.h:106:31: error: expected identifier or ‘(’ before ‘[’ token
     __syscall_slong_t __unused[3];
                               ^
/usr/include/bits/stat.h:164:31: error: expected identifier or ‘(’ before ‘[’ token
     __syscall_slong_t __unused[3];
                               ^
gcc  -std=gnu99 -DHAVE_CONFIG_H -I. -I../git  -I../git/include -I../git -include config.h -isystem/data/meta-iot2000/build/tmp/work/x86_64-linux/efibootguard-native/0.7-r0/recipe-sysroot-native/usr/include -Wno-unused-parameter -Wmissing-prototypes -fshort-wchar -DHAVE_ENDIAN_H -D_GNU_SOURCE -D_FILE_OFFSET_BITS=64 -Werror -isystem/data/meta-iot2000/build/tmp/work/x86_64-linux/efibootguard-native/0.7-r0/recipe-sysroot-native/usr/include -O2 -pipe -c -o tools/bg_setenv-bg_setenv.o `test -f 'tools/bg_setenv.c' || echo '../git/'`tools/bg_setenv.c
make: *** [tools/libebgenv_a-ebgpart.o] Error 1
make: *** Waiting for unfinished jobs....
WARNING: /data/meta-iot2000/build/tmp/work/x86_64-linux/efibootguard-native/0.7-r0/temp/run.do_compile.14380:1 exit 1 from 'exit 1
`

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal-4.8"
TARGET_SYS           = "i586-poky-linux"
MACHINE              = "iot2000"
DISTRO               = "poky-iot2000"
DISTRO_VERSION       = "V3.1.1"
TUNE_FEATURES        = "m32 i586-nlp"
TARGET_FPU           = ""
meta-efibootguard    = "HEAD:e13f038bb362a6c086be50359f1bf31cf770e151"
meta-iot2000-bsp     
meta-iot2000-example = "master:9c71919268f79ced2f0c3c823dd3b9f7b077eef0"
meta-networking      
meta-oe              
meta-python          = "HEAD:cc6fc6b1641ab23089c1e3bba11e0c6394f0867c"
meta-swupdate        = "HEAD:1886350dacb63d931b3d1a3809b27795b0a5c306"
meta                 
meta-poky            
meta-yocto-bsp       = "HEAD:febbe2944c0c4a04b85fa98fdc261186115954d8"

DEBUG: Executing python function autotools_aclocals
DEBUG: SITE files ['endian-little', 'common-linux', 'common-glibc', 'bit-64', 'x86_64-linux', 'common']
DEBUG: Python function autotools_aclocals finished
DEBUG: Executing shell function do_compile

Linux devel_buildvm01_cos7 3.10.0-1160.24.1.el7.x86_64 #1 SMP Thu Apr 8 19:51:47 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

glibc-headers-2.17-323.el7_9.x86_64 : Header files for development using standard C libraries.
Repozitář    : @updates
Shoda s:
Soubor      : /usr/include/bits/stat.h

What am I doing wrong ?

Thank you for answer.

Best Regards
Marek Matejka

jan-kiszka, 2021-04-29:
How did you invoke the build (command line)?

matejkamarek, 2021-04-29:
> How did you invoke the build (command line)? And, if not using the kas container, what is your build environment?

Yes:
$ kas build kas-example.yml 
$ printenv 
XDG_SESSION_ID=20
HOSTNAME=devel_buildvm01_cos7
TERM=xterm-256color
SHELL=/bin/bash
HISTSIZE=1000
SSH_CLIENT=192.168.122.1 57002 22
OLDPWD=/home/maty/meta-iot2000/poky
GIT_SSL_CAINFO=/home/maty/meta-iot2000/poky/buildtools/sysroots/x86_64-pokysdk-linux/etc/ssl/certs/ca-certificates.crt
SSH_TTY=/dev/pts/0
USER=maty
LS_COLORS=rs=0:di=38;5;27:ln=38;5;51:mh=44;38;5;15:pi=40;38;5;11:so=38;5;13:do=38;5;5:bd=48;5;232;38;5;11:cd=48;5;232;38;5;3:or=48;5;232;38;5;9:mi=05;48;5;232;38;5;15:su=48;5;196;38;5;15:sg=48;5;11;38;5;16:ca=48;5;196;38;5;226:tw=48;5;10;38;5;16:ow=48;5;10;38;5;21:st=48;5;21;38;5;15:ex=38;5;34:*.tar=38;5;9:*.tgz=38;5;9:*.arc=38;5;9:*.arj=38;5;9:*.taz=38;5;9:*.lha=38;5;9:*.lz4=38;5;9:*.lzh=38;5;9:*.lzma=38;5;9:*.tlz=38;5;9:*.txz=38;5;9:*.tzo=38;5;9:*.t7z=38;5;9:*.zip=38;5;9:*.z=38;5;9:*.Z=38;5;9:*.dz=38;5;9:*.gz=38;5;9:*.lrz=38;5;9:*.lz=38;5;9:*.lzo=38;5;9:*.xz=38;5;9:*.bz2=38;5;9:*.bz=38;5;9:*.tbz=38;5;9:*.tbz2=38;5;9:*.tz=38;5;9:*.deb=38;5;9:*.rpm=38;5;9:*.jar=38;5;9:*.war=38;5;9:*.ear=38;5;9:*.sar=38;5;9:*.rar=38;5;9:*.alz=38;5;9:*.ace=38;5;9:*.zoo=38;5;9:*.cpio=38;5;9:*.7z=38;5;9:*.rz=38;5;9:*.cab=38;5;9:*.jpg=38;5;13:*.jpeg=38;5;13:*.gif=38;5;13:*.bmp=38;5;13:*.pbm=38;5;13:*.pgm=38;5;13:*.ppm=38;5;13:*.tga=38;5;13:*.xbm=38;5;13:*.xpm=38;5;13:*.tif=38;5;13:*.tiff=38;5;13:*.png=38;5;13:*.svg=38;5;13:*.svgz=38;5;13:*.mng=38;5;13:*.pcx=38;5;13:*.mov=38;5;13:*.mpg=38;5;13:*.mpeg=38;5;13:*.m2v=38;5;13:*.mkv=38;5;13:*.webm=38;5;13:*.ogm=38;5;13:*.mp4=38;5;13:*.m4v=38;5;13:*.mp4v=38;5;13:*.vob=38;5;13:*.qt=38;5;13:*.nuv=38;5;13:*.wmv=38;5;13:*.asf=38;5;13:*.rm=38;5;13:*.rmvb=38;5;13:*.flc=38;5;13:*.avi=38;5;13:*.fli=38;5;13:*.flv=38;5;13:*.gl=38;5;13:*.dl=38;5;13:*.xcf=38;5;13:*.xwd=38;5;13:*.yuv=38;5;13:*.cgm=38;5;13:*.emf=38;5;13:*.axv=38;5;13:*.anx=38;5;13:*.ogv=38;5;13:*.ogx=38;5;13:*.aac=38;5;45:*.au=38;5;45:*.flac=38;5;45:*.mid=38;5;45:*.midi=38;5;45:*.mka=38;5;45:*.mp3=38;5;45:*.mpc=38;5;45:*.ogg=38;5;45:*.ra=38;5;45:*.wav=38;5;45:*.axa=38;5;45:*.oga=38;5;45:*.spx=38;5;45:*.xspf=38;5;45:
MAIL=/var/spool/mail/maty
PATH=/home/maty/meta-iot2000/poky/buildtools/sysroots/x86_64-pokysdk-linux/usr/bin:/home/maty/meta-iot2000/poky/buildtools/sysroots/x86_64-pokysdk-linux/usr/bin:/home/maty/meta-iot2000/poky/buildtools/sysroots/x86_64-pokysdk-linux/usr/bin:/usr/lib64/ccache:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin:/home/maty/.local/bin:/home/maty/bin
PWD=/home/maty/meta-iot2000
XMODIFIERS=@im=none
LANG=cs_CZ.UTF-8
HISTCONTROL=ignoredups
SHLVL=1
HOME=/home/maty
OPENSSL_CONF=/home/maty/meta-iot2000/poky/buildtools/sysroots/x86_64-pokysdk-linux/etc/ssl/openssl.cnf
LOGNAME=maty
SSH_CONNECTION=192.168.122.1 57002 192.168.122.199 22
LESSOPEN=||/usr/bin/lesspipe.sh %s
XDG_RUNTIME_DIR=/run/user/1000
_=/usr/bin/printenv

Matejka

jan-kiszka, 2021-04-29:
And what was your host OS? Did you install all required packages there for building Yocto?

Does `kas-container build kas-example.yml` work (if you enable docker for the building user)? Or did you run that kas command inside kas:2.4 in fact? I would just like to make sure we rule out external problems.

I'm currently retrying `kas-container build kas-example.yml --target efibootguard-native` on a clean checkout.

jan-kiszka, 2021-04-29:
Works fine here. Also "native" build on an OpenSUSE 15.2 with required dependencies installed.

I suspect the issue is related to your host OS, its age or its package set. The error comes from a system header that efibootguard only includes.

matejkamarek, 2021-04-29:
> And what was your host OS? Did you install all required packages there for building Yocto?
>
> Does `kas-container build kas-example.yml` work (if you enable docker for the building user)? Or did you [...]

Build OS is CentOS Linux release 7.9.2009 running via qemu-system-x86_64 on host OS Fedora release 32.

Kas install in build os centos via:

$ pip3 install kas --user

This way, the image cannot be assembled ?

jan-kiszka, 2021-04-29:
Yeah, just spotted the kernel version, currently trying to reproduce in a centos:7 container.

Note that, in order to build natively, you need to install more than kas into that system. Yocto lists the native dependencies. You can avoid that by using the kas container which has everything needed pre-installed (based on Debian 10).

jan-kiszka, 2021-04-29:
Centos7 is too old:
```
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    Your version of gcc is older than 5.0 and will break builds. Please install a newer version of gcc (you could use the project's buildtools-extended-tarball or use scripts/install-buildtools).
Your version of tar is older than 1.28 and does not have the support needed to enable reproducible builds. Please install a newer version of tar (you could use the project's buildtools-tarball from our last release or use scripts/install-buildtools).
```

matejkamarek, 2021-04-29:
> Centos7 is too old:
>
> ```
> ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
>     Either fix the cause of this error or at your own risk disable the checker (see [...]

$ cat /etc/redhat-release 
Fedora release 32 (Thirty Two)
$ docker -v
Docker version 20.10.6, build 370c289
$ kas-container build meta-iot2000/kas-bsp.yml

This way is OK ?

Thank you.

Matejka

jan-kiszka, 2021-04-29:
That should be fine.

matejkamarek, 2021-04-30:
> That should be fine.

Hi,

build succeeded.

Thank you very much.

Matejka

closed 2021-04-30 by matejkamarek