siemens/meta-iot2000 #171: Writing DQs with node-red-node-intel-gpio causes constant output of 1

BrotKolben, 2021-08-02:
Dear colleagues,
As described on the [IOT2000 Forum](https://support.industry.siemens.com/tf/ww/en/posts/266189/) there is an issue regarding setting the digital outputs with node-red using the preinstalled nodes. The provided flow seems to be working fine upon start/reboot of the iot2040. As soon as a further "deploy" is done in node-red, the error can be triggered. In this case the digital outputs are then constantly set to 1 and never return to 0.
To reproduce, use the example Image V3.1.1 (available on SIOS) together with the flow provided in attachment.  
[flows (9).zip](https://github.com/siemens/meta-iot2000/files/6917191/flows.9.zip). 
In node-red, The debug window shows changing values 0 and 1 every 2 seconds: 
[image]
Wire e.g Digital Output 0 directly to Digital Input 0 on the I/O Shield, or use a secondary device with a DI. 
When reading the Digital Input 0 in a loop using mraa, the value will be changing between 0 and 1 as intended (example code available in attachment.) [read-di-test.zip](https://github.com/siemens/meta-iot2000/files/6917248/read-di-test.zip)
[image]
After re-deploying the flow in node-red, the value will remain at 1 constantly. 
[image]

BR

BrotKolben, 2021-08-02:
Recommended customer to try with a different delay time and error appears to be gone for me at least... Waiting for customer feedback. Will leave the issue open until we get a response...

BR

jan-kiszka, 2021-08-02:
Sounds like https://github.com/siemens/meta-iot2050/issues/57 -> https://github.com/siemens/meta-iot2000/issues/166. Didn't make it into the last release IIRC.