siemens/meta-iot2000 #173: Pre-installed Modbus node is not compatible and leads to crash of Node-Red

bergmanu, 2021-12-07:
Pre-installed node-red-contrib-modbus v5.13.3 is not compatible to NodeJS 8 and Node-Red V1.x (see [here](https://www.npmjs.com/package/node-red-contrib-modbus)). 
A contributer in the forum came across this and solved the issue by unstalling the node and re-install V3.6.1, which is compatible to NodeJS 8 and Node-Red V1.x (see [here](https://support.industry.siemens.com/tf/ww/en/posts/modbus-tcp-crash-my-node-red-in-simatic-iot2040/272722/?page=0&pageSize=10)).

bergmanu, 2022-02-16:
can this be solved with the (maybe) upcoming image version?

jan-kiszka, 2022-02-16:
Is 3.6.1 the last supported version for NodeJS 8? Yes, we should be able to resolve that.

bergmanu, 2022-02-16:
Yes, according to [this list](https://flows.nodered.org/node/node-red-contrib-modbus) and the [available releases](https://github.com/BiancoRoyal/node-red-contrib-modbus/releases?page=2) on github, 3.6.1 is the latest release working with nodeJS 8. 
Using this version 3.6.1, node-red version 0.x is required. Not sure which version is planned to be integrated in the next image.

jan-kiszka, 2022-02-16:
That plan is to move as much up with Node RED as its support for Node JS 8 permits.