siemens/meta-iot2000 #175: Not possible to install nodes with V3.1.1

bergmanu, 2022-02-18:
Since some time it is not possible any more to install any node with the Example Image V3.1.1. 
No matter which node is installed this error appears: 
[image]

I tried to update fsevents, which fails with this error: 
[image]

jan-kiszka, 2022-02-18:
Looks as if that fsevents packages got broken upstream - if it only wants to work on Mac.

bergmanu, 2022-02-18:
Oh right, didn't recognize the MAC thing.

bergmanu, 2022-02-24:
I did some further tests. Seems that installing node globally (-g) works, but installing via Manage Palette doesn't work. 

Test steps with s7 node: 
- npm install -g node-red-contrib-s7@2.1.1 --> works
- Update node-red-contrib-s7 to V3.0 via Manage Palette --> fails with fsevents error
- npm install -g node-red-contrib-s7 --> works and installs version 3.1
- npm install -g node-red-contrib-s7@3.0 --> works

Test steps with bit node: 
- Install node-red-contrb-bit via Manage Palette --> fails with fsevents error
- npm install -g node-red-contrib-bit --> works

Test steps with config node: 
- Install node-red-contrb-config via Manage Palette --> no response
- npm install -g node-red-contrib-config--> works

closed 2022-05-18 by bergmanu